johanmeijer/grott #651: Grott starts day with total production from previous day

bvkrieken, 2025-04-29:
Hi there, 

I'm experiencing a frustrating issue that Grott very often starts the day with the total production of the previous day, completely ruining the statistics and graphs. It doesn't always do this, there are days when things are just fine. What could cause this issue?

[image]
[image]
[image]

johanmeijer, 2025-04-30:
Hmm strange. Grott does not  changed the values it only passes what the inverter is sending, so probably this is an faulty first data record the inverter sends. 

Can you send the first record (xxxx04) of the day then maybe we can find out what is wrong. 

Do you also use grottserver of are you sending the data also to Growatt?

Supered1967, 2025-04-30:
I had same kind of issue in the past. At the end the dongle (shine-WiFi-s) was faulty. I replaced it with a new one and the issue was gone.
(For some reason the old was updated from Growatt with a firmware for a different dongle)

Regards
Ed

marcblokland, 2025-05-14:
> Today it is even worse... Now it seems it's started the day with the total production of this year. Grott used to work just fine a few weeks ago and ran without problems for at least half a year, what changed? I didn't change anything in the configuration or my pv installation.
> 
> [image]

have the exact same issue

johanmeijer, 2025-05-14:
I have not changed anything in Grott 2.8.3 either. I think indeed that it might be the datalogger (wifi dongle). I have seen dongle strange behavior because of time drift. 

Is your inverter always on or does it stop working in the evening and starts again in the morning. 

What you can try to do to specify server=server in the grott.ini (or for docker : gtime=server). This will force that grott uses the system time in the MQTT message in stead of the datalogger time.

dberrocal-git, 2025-05-16:
Just out of curiosity, what is the metrics update time? I remember that when you go below 0.5minutes the data logger gives problems.

marcblokland, 2025-05-16:
1 minute. Firmware update did not solve the problem. 

Looks like the all Pv today total sensor is causing the problem.

johanmeijer, 2025-05-21:
Can you send a data record (the first one in the morning) from the Grott log then we can at least determine if the data record is wrong or that there is something going on in HA.

marcblokland, 2025-05-24:
> Can you send a data record (the first one in the morning) from the Grott log then we can at least determine if the data record is wrong or that there is something going on in HA.

 -  pvpowerout           :  0.0
		 -  pvfrequentie         :  0.0
		 -  pvgridvoltage        :  0.0
		 -  pvgridcurrent        :  0.0
		 -  pvgridpower          :  0.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  70982541.8
		 -  totworktime          :  145.6
		 -  pvenergytoday        :  24189739.4
		 -  pvenergytotal        :  26050960.8
		 -  epvtotal             :  18.8
		 -  epv1today            :  4244.9
		 -  epv1total            :  576387.7
		 -  epv2today            :  10.5
		 -  epv2total            :  2555.9
		 -  pvtemperature        :  57.0
		 -  pvipmtemperature     :  0.0
	 - MQTT jsonmsg: 
			 {"device": "RLHDCK900Y", "time": "2025-05-24T15:32:53", "buffered": "no",
			 "values": {"datalogserial": "GPG0CG70W7", "pvserial": "RLHDCK900Y",
			 "pvstatus": 1, "pvpowerin": 10995, "pv1voltage": 6284, "pv1current": 9,
			 "pv1watt": 5655, "pv2voltage": 6145, "pv2current": 8, "pv2watt": 4916,
			 "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0,
			 "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2":
			 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 709825418,
			 "totworktime": 1048576, "pvenergytoday": 241897394, "pvenergytotal":
			 260509608, "epvtotal": 188, "epv1today": 42449, "epv1total": 5763877,
			 "epv2today": 105, "epv2total": 25559, "pvtemperature": 570,
			 "pvipmtemperature": 0}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 
	 - Grottserver - Data received from : 127.0.0.1:57138
	 - Grottserver - Original Data:
		 \x00\x5f\x00\x06\x02\x41\x01\x04\x00\x22\x28\x47\x22\x33\x43\x77\x25\x58\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3d\x3b\x29\x30\x37\x0c\x4b\x5f\x47\x38\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\x61\x5e\x87\x5f\xfe\x6f\x7e\x61\x74\x62\x50\x6a\x6e
		 \x77\x69\x74\x74\x54\x46\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x5d
		 \x2e\x67\xfe\x4e\x7f\x6f\x67\x61\x74\x7a\x3c\x7b\x6c\x77\x71\x74\x74\x49\x19
		 \x66\x74\x61\x64\x74\x47\x7c\x04\x78\xd3\x7b\xf3\x48\xda\x6f\x77\x61\xc8\x74
		 \x47\xd7\xbe\x77\x36\x87\x51\x47\x72\x6f\x1e\x61\x74\x17\x90\x72\x6f\x77\x35
		 \x74\x74\x07\x97\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xd3\xdd\x76
		 \x4e\x46\x4a\x6e\x68\x61\x74\x74\x47\x7e\x8e\x7b\xdf\x3a\x54\x47\x72\x6f\x76
		 \xcc\xc4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x68\x74\x48\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x76\x23\x3e\xd7\x74\x74\x47\x72\x6f\x6e\x61\x71\x74\x50\x72\x76\x77\x64
		 \x74\x6c\x47\x72\x6f\x77\x61\x74\x33\x62\x72\x6d\x75\x41\x76\x6a\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
[... 6 lines cut ...]
		 24c4844434b393030590000000000000000000000000000000000000000000000000000020000
		 007c000100002af3188c000900001617180100080000133400000000000000000000000000000
		 000000000000000000000000000000000000000000000000000000000000000000000002a4f13
		 8a090d001000000e7b0903001000000e6b0903001000000e6b0fb20f870fa8000000bc0000a5d
		 10057f32500000069000063d700000054000040e5000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000000000000a4bc023a01380
		 11f000000000ce10cbe4e2000000001adb0000000000000000000000000000000000000000900
		 3c0000000000000000000000000000000000000000007d00f9000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000000000000000044c49b600
		 000000001900050017001900050018000000000000472500020220021e0000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000029db0000007e0000d021000000000000000000000000000000000000000000000
		 000000000009be9
	 - Grottserver - 0104 data record received
	 - Grottserver - Response: 
[... cut ...] 0\x06\x00\x03\x01\x04\x47\x9c\x5c
	 - Grottserver - Put response on queue:  127.0.0.1_57138  msg: 
		 \x00\x5f\x00\x06\x00\x03\x01\x04\x47\x9c\x5c
	 - Grottserver - get response from queue:  127.0.0.1_57138 msg: 
		 \x00\x5f\x00\x06\x00\x03\x01\x04\x47\x9c\x5c
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.3', 5279), raddr=('172.17.0.1', 40048)>
	 - Data less then minimum record length, data not processed