johanmeijer/grott #652: ShineX Adapter doesn't connect to Grott

TrueMB, 2025-04-29:
Hello!

Thank you very much for that project. It really helped me for my Smarthome!
Also, this issue has nothing to do with grott itself, but more the connection to it.

Sadly I misconfigured my HomeNetwork. I was using the range: 172.5.5.0/24, which is a public IP range. (That did work though)
So I changed it to a correct private Network.

Growatt Shine X Adapter -> my docker server with Grott Proxy -> Growatt Online Servers

Since my Adapter was pointing to my wrong old IP, I changed it back to the Online IP Address.
After that I got my information online through the App.

I tried setting the IP Online through the Webinterface back to my local IP Address of my Server, but it always added a "66" to the Address, which made it wrong.

Then I tried changing the IP directly on the Adapter setup advance menu, like I did before for the online IP Address. Now it also seems to have disconnected again from the growatt cloud, but I don't see any traffic on the console for grott. I have no issues connecting to the server and port with Windows (powershell: Test-NetConnection -ComputerName XXX -Port 5279). 

Since I also don't see traffic directly on the firewall, it seems like the Adapter isn't sending any data.

Does anyone have an idea, what I could test or what I might be doing wrong?

tabascoz, 2025-04-29:
> I tried setting the IP Online through the Webinterface back to my local IP Address of my Server, but it always added a "66" to the Address, which made it wrong.

What do you mean by that, I don't think i get it. When you changed in the webinterface the final IP was being suffixed by "66"? It's 
very strange... 

Is your home network now using a common range, let's say 192.168.0.0 netmask 255.255.255.0?  Also if you could give more information of it just to rule out any issues... 

my two cents, would  recommend using Growatt  mobile application to set the parameters, @johanmeijer  had some issues with Growatt web interface, it may help. 

Cheers, Tabascoz

TrueMB, 2025-04-29:
Thank you very much for your time, reading into this!

I mean the Webinterface of server.growatt.com
I set the IP using the "Set IP" Option and with register 17. But everytime I read the value, it shows the set local IP address + 66.
For example: 192.168.0.1066 (192.168.0.10 would be the docker server)
My Adapters are currently offline, so I cant send a complete screenshot. But this is, where I tried it first:
[image]

The Growatt Adapter are in a 192.168.0.0/24 Network and the DMZ in 10.5.5.0/24 (slightly changed to the real ones)
But like I said, no firewall traffic out of the growatt adapter. Not into the WAN and not into the DMZ.
My HomeNetwork is also in another different network. My PC successfully creates a response in the grott console.
So it has to be something with the adapters.....

I did try it with the data logger configuration tool from shinephone. Since then it stopped the connection to the cloud and it also seems to have correctly set the IP. But no output is generated...

johanmeijer, 2025-04-30:
Set IP works fine for me. 

But I remember that when you initialized the shinewifi-x the first time (via the mobile app or directly connecting to the stick) you can specify if you want to use IP or Domen (URL). Maybe that switch (register?) is set and he does not use the IP address? 

I am just guessing.

TrueMB, 2025-05-01:
Thank you very much for your ideas. 

I think I managed it to make it work again. 
It seems, that this option on my wireless accesspoint was conflicting with something.
[image]

Since I didn't change that value, I think the issue was, that I was moving from a public IP range to a private IP range (which was blocked to the guest option)

Maybe the accesspoint allowed the traffic, since it didn't count as a private network.

closed 2025-05-01 by TrueMB