johanmeijer/grott #656: MIC?

GherardS, 2025-05-09:
I have a 3000tlx microphone that is not compatible with grott?

GherardS, 2025-05-09:
Yes, I have problems!
I have homeassistant on VM in Proxmox, I put ip on shineLanbox of the homeassistant host but I don't receive the data…

GherardS, 2025-05-10:
Grott Growatt logging monitor : 2.8.3
Grott process configuration file
Grott process environmental variables
	Pulled 'ginvtype=MIC' from the environment
	Pulled 'gnomqtt=False' from the environment
	Pulled 'gmqttip=172.30.33.0' from the environment
	Pulled 'gmqttport=1883' from the environment
	Pulled 'gmqttauth=True' from the environment
	Pulled 'gmqttuser=addons' from the environment
	Pulled 'gmqttpassword=Aoboan7phipeiWo6ohrooch5Wi9ui9oe0Keeg0niefohro2mae4aireiQuae2que' from the environment
Grott override settings if set in commandline
Grott process json layout files
Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott-beta
IP :  172.30.33.4 , port :  5279 
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.4', 5279), raddr=('192.168.15.70', 1029)>
	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.4', 5279), raddr=('192.168.15.70', 1029)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51704), raddr=('47.91.67.66', 5279)>

This is my log

johanmeijer, 2025-05-11:
Hmm I think data is received. Otherwise you would not see the Package information. But I do not see any data records and that is strange. Do you receive any information in the growatt website or phone app?

I am not totally up to speed with the HA plugin (I think you use that one). Do you now if it possible to start with verbose parameter (then we get more information).

tabascoz, 2025-05-12:
> I am not totally up to speed with the HA plugin (I think you use that one). Do you now if it possible to start with verbose parameter (then we get more information).

I use it. The plugin relies on MQTT to get data inside HA. So if MQTT is enabled, and you can  see messages from grott there and also MQTT is working in HomeAssistant, it should work.

orson74, 2025-10-21:
Is there anybody who has a solution?