johanmeijer/grott #662: Incorrect data decryption for SPF-5000-ES

bastidmn, 2025-05-25:
Hey,
first of all, thank you for this amazing project, the setup and connection to my HA went without issue.

But when setting up my SPF-5000-ES inverter, the data parsing seems to be incorrect, even though some values are recognized fine, most fields are not matching what the ShinePhone app and the inverter itself are reporting.

batterySoc, bat_Volt, BatWatt/BatDischarVA/BatDischarWatt, vpv1, ppv1, ebatDischarToday/Total, epvtoday/total, pvenergytoday, datalogserial and pvserial (and possibly a few I've missed) seem to match.

Here are the logs for one data package:
```
grott    |       - Growatt original Data:
grott    |               \x00\x56\x00\x06\x01\x5f\x01\x04\x03\x36\x2b\x47\x22\x3e\x37\x75\x44\x59\x77
grott    |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott    |               \x25\x39\x2a\x45\x37\x0d\x39\x5e\x40\x50\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott    |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x78\x71\x6d\x54\x74\x5c\x74\x61
grott    |               \x74\x74\x6b\x72\x63\x79\x05\x74\x74\x47\x72\x62\x0d\x61\x74\x74\x47\x72\x2e
grott    |               \x77\x61\x74\x74\x50\xfc\x6f\x77\x7b\x56\x74\x47\x72\x6f\x77\x61\x74\x74\x53
grott    |               \xc0\x6f\x2e\x71\xc7\x74\x47\x72\x6f\x7f\x9d\x67\xfc\x47\x72\x6f\xa3\x61\xb1
grott    |               \x74\xc2\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xd6\x61\xbc\x74\x5a\x72\x4d\x77
grott    |               \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x3a\xfd\x47\x5f
grott    |               \x6f\x2e\x68\x5c\x74\x47\x72\x6f\x77\x61\x74\x22\x47\x72\xb5\x8f\x61\x74\x74
grott    |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x64\x8a\x74\x47\x72\x73\x77\x61
grott    |               \x28\x2b\x47\x72\x6f\x77\x61\x74\x25\xeb\x72\x6f\x77\x61\x6c\x72\x47\x72\x75
grott    |               \xd9\x61\x74\x7f\xb9\x72\x6f\x7c\x9f\x74\x74\x4c\x8c\x6f\x77\x61\x74\x74\x47
grott    |               \x72\x4f\x77\x61\x74\x4e\x47\x72\x6f\x3e\x61\x74\x6b\xdb\x72\x6f\x77\x3b\x74
grott    |               \xf2\x47\x72\x6f\x76\x35\x5b\x7c\xe2\x61\xe8\x62\x9d\x67\x25\x55\x9c\x67\xeb
grott    |               \x72\xf8\x74\x57\x72\x66\x77\x65\x74\x65\x47\x71\x6f\x9e\x61\xa3\x74\xa0\x72
grott    |               \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott    |               \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott    |               \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\xd3\x3e
grott    |       - Grott automatic protocol detection
grott    |       - Grott data record length 359
grott    |       - layout   :  T060104SPF
grott    |       - no matching record layout found, try generic
grott    |       - Record layout used :  T06NNNNSPF
grott    |       - Growatt data decrypted V2
grott    |       - Grott Growatt data decrypted
grott    |       - Growatt plain data:
grott    |               00560006015f010444444430434a4332363600000000000000000000000000000000000000004
grott    |               a4e4b31434a4b3137310000000000000000000000000000000000000000190519130633030000
grott    |               002c000c0e64000000000d7a00000000004100000000178e00001a22000000000000000014b20
grott    |               05910b30000000008fc1388000000d400c50085000000000000000000a100c8001d0022000000
grott    |               000000000000000000000000004e89002d0059092800000000000000560000daf800000000000
grott    |               0000000000000000005fe0000001c00005c5f00000000000051ac00000000180600001aae0000
grott    |               0bfe00000bfe00000bfe00000000000000200000003a0000004900001f9c0000005a008600000
grott    |               001542f08a5138715fc135112ee089c138c0010000900040011000300e900d700e70000000000
grott    |               00000000000000000000000000000000000000000000000000000000000000000000000000000
grott    |               000000000000000000000a779
grott    |       - Growatt new layout processing
grott    |               - decrypt       :  True
grott    |               - offset        :  6
grott    |               - record layout :  T06NNNNSPF
grott    |
grott    |       - Grott data record date/time processing started
grott    |       - date-time:  2025-05-25T19:06:51
grott    |       - Grott values retrieved:
grott    |               -  datalogserial        :  DDD0CJC266
grott    |               -  pvserial             :  JNK1CJK171
grott    |               -  pvstatus             :  12
grott    |               -  vpv1                 :  368.4
grott    |               -  vpv2                 :  0.0
grott    |               -  ppv1                 :  345.0
grott    |               -  ppv2                 :  0.0
grott    |               -  buck1curr            :  6.5
grott    |               -  buck2curr            :  0.0
grott    |               -  op_watt              :  603.0
grott    |               -  pvpowerout           :  603.0
grott    |               -  op_va                :  171264.0
grott    |               -  acchr_watt           :  0.0
grott    |               -  acchr_VA             :  0.0
grott    |               -  bat_Volt             :  53.0
[... 36 lines cut ...]
grott    |       - MQTT jsonmsg:
grott    |                       {"device": "JNK1CJK171", "time": "2025-05-25T19:06:51", "buffered": "no",
grott    |                       "values": {"datalogserial": "DDD0CJC266", "pvserial": "JNK1CJK171",
grott    |                       "pvstatus": 12, "vpv1": 3684, "vpv2": 0, "ppv1": 3450, "ppv2": 0,
grott    |                       "buck1curr": 65, "buck2curr": 0, "op_watt": 6030, "pvpowerout": 6030,
grott    |                       "op_va": 1712640, "acchr_watt": 0, "acchr_VA": 0, "bat_Volt": 5298,
grott    |                       "batterySoc": 89, "bus_volt": 4275, "grid_volt": 0, "line_freq": 0,
grott    |                       "outputvolt": 2300, "pvgridvoltage": 2300, "outputfreq": 5000, "invtemp":
grott    |                       212, "dcdctemp": 197, "loadpercent": 133, "buck1_ntc": 161, "buck2_ntc":
grott    |                       200, "OP_Curr": 29, "Inv_Curr": 34, "AC_InWatt": 0, "AC_InVA": 0,
grott    |                       "faultBit": 0, "warningBit": 0, "faultValue": 0, "warningValue": 0,
grott    |                       "constantPowerOK": 2344, "epvtoday": 86, "pvenergytoday": 86, "epvtotal":
grott    |                       56056, "eacCharToday": 0, "eacCharTotal": 1534, "ebatDischarToday": 28,
grott    |                       "ebatDischarTotal": 23647, "eacDischarToday": 0, "eacDischarTotal": 20908,
grott    |                       "ACCharCurr": 0, "ACDischarWatt": 6150, "ACDischarVA": 6830,
grott    |                       "BatDischarWatt": 3070, "BatDischarVA": 3070, "BatWatt": 3070}}
grott    |       - Grott MQTT topic used : energy/growatt
grott    |       - MQTT message message sent
grott    |       - Grott Send data to PVOutput disabled
grott    |       - Grott Send data to Influx disabled
grott    |       - Grott extension processing disabled
```
In my docker-compose, I've set ginvtype to spf and I also tried a few other invtypes, but - at least from the ones I tried - no values more fitting than this were produced.

I would be very thankful if you/anyone could help me get the correct readings!

tabascoz, 2025-05-25:
Hi there, i have the exact same inverter. from the log it looks fine. It occurred me two things: 

-  you may be receiving historical records, check the time if it matches the current time, or 
- Growatt information is wrong, check in inverter's LCD and see if the values matches or, 

Here's one sample of the data i receive with Grott 2.8: 

```
[{'measurement': 'EUH5CB9037', 'time': '2025-05-25T18:34:23', 'fields':
May 25 15:34:29 pro grott[901156]:                       {'datalogserial': 'MWG0BH7030', 'pvserial': 'EUH5CB9037', 'pvstatus': 12,
May 25 15:34:29 pro grott[901156]:                       'vpv1': 2137, 'vpv2': 0, 'ppv1': 1460, 'ppv2': 0, 'buck1curr': 29,
May 25 15:34:29 pro grott[901156]:                       'buck2curr': 0, 'op_watt': 1240, 'pvpowerout': 1240, 'op_va': 501760,
May 25 15:34:29 pro grott[901156]:                       'acchr_watt': 0, 'acchr_VA': 0, 'bat_Volt': 5030, 'batterySoc': 50,
May 25 15:34:29 pro grott[901156]:                       'bus_volt': 4027, 'grid_volt': 2270, 'line_freq': 5996, 'outputvolt': 2334,
May 25 15:34:29 pro grott[901156]:                       'pvgridvoltage': 2334, 'outputfreq': 6000, 'invtemp': 217, 'dcdctemp': 211,
May 25 15:34:29 pro grott[901156]:                       'loadpercent': 24, 'buck1_ntc': 207, 'buck2_ntc': 217, 'OP_Curr': 8,
May 25 15:34:29 pro grott[901156]:                       'Inv_Curr': 21, 'AC_InWatt': 0, 'AC_InVA': 0, 'faultBit': 0, 'warningBit':
May 25 15:34:29 pro grott[901156]:                       0, 'faultValue': 0, 'warningValue': 0, 'constantPowerOK': 2510, 'epvtoday':
May 25 15:34:29 pro grott[901156]:                       20, 'pvenergytoday': 20, 'epvtotal': 36394, 'eacCharToday': 5,
May 25 15:34:29 pro grott[901156]:                       'eacCharTotal': 6818, 'ebatDischarToday': 1, 'ebatDischarTotal': 6559,
May 25 15:34:29 pro grott[901156]:                       'eacDischarToday': 22, 'eacDischarTotal': 50475, 'ACCharCurr': 0,
May 25 15:34:29 pro grott[901156]:                       'ACDischarWatt': 1290, 'ACDischarVA': 1960, 'BatDischarWatt': 0,
May 25 15:34:29 pro grott[901156]:                       'BatDischarVA': 0, 'BatWatt': 0, 'Fan1': 24}}]
```

looks very much the same. Nowadays to be honest i trust Grott data by far more than what comes from Growatt sw. Btw it's been over a year that i don't use their sw anyway. 

Let me know the checks and happy to support.

bastidmn, 2025-05-26:
Hey, thanks to both of you for your answers!
First of all, I restarted Grott with the gtime="server" parameter and now the reports definitely match up regarding their actuality.
The main reason why I opened this issue was, that I have a (calibrated) wall socket for some utilities connected to the inverter's circuit and multiples times the power draw reported from the socket was higher than the (reported) power output of the inverter, which just doesn't seem to add up. 
I just want to check if I failed to configure something correctly, i.e. the MQTT values I parse in HA:

- Should I use op_watt/pvpowerout as overall power output of the whole system (which I did when these values where reported)
- What are ACDischarWatt/VA, eacCharToday/Total and eacDisCharToday/Total?

Did I mess up the Grott configuration currently running?
```
grott    | Grott settings:
grott    |
grott    | _Generic:
grott    |      version:                 2.7.8
grott    |      verbose:                 True
grott    |      trace:                   False
grott    |      config file:             grott.ini
grott    |      minrecl:                 100
grott    |      decrypt:                 True
grott    |      compat:                  False
grott    |      invtype:                 spf
grott    |      invtypemap:              {}
grott    |      include_all:             False
grott    |      blockcmd:                False
grott    |      noipf:                   False
grott    |      time:                    auto
grott    |      sendbuf:                 True
grott    |      timezone:                local
grott    |      valueoffset:             6
grott    |      offset:                  6
grott    |      inverterid:              automatic
grott    |      mode:                    proxy
grott    |      grottip                  default
grott    |      grottport                5279
grott    | _MQTT:
grott    |      nomqtt                   False
grott    |      mqttip:                  192.168.178.164
grott    |      mqttport:                1883
grott    |      mqtttopic:               energy/growatt
grott    |      mqttmtopic:              False
grott    |      mqttmtopicname:          energy/meter
grott    |      mqttinverterintopic:     False
grott    |      mqtttretain:             False
grott    |      mqtttauth:               True
grott    |      mqttuser:                grott
grott    |      mqttpsw:                 **secret**
grott    | _Growatt server:
grott    |      growattip:               47.91.67.66
grott    |      growattport:             5279
grott    | _PVOutput:
grott    |      pvoutput:                False
grott    |      pvdisv1:                 False
grott    |      pvtemp:                  False
grott    |      pvurl:                   https://pvoutput.org/service/r2/addstatus.jsp
grott    |      pvapikey:                yourapikey
grott    |      pvinverters:             1
grott    |      pvsystemid:              12345
grott    | _Influxdb:
grott    |      influx:                  False
grott    |      influx2:                 False
grott    |      database:                grottdb
grott    |      ip:                      localhost
grott    |      port:                    8086
grott    |      user:                    grott
grott    |      password:                **secret**
grott    |      organization:            grottorg
grott    |      bucket:                  grottdb
grott    |      token:                   **secret**
grott    | _Extension:
grott    |      extension:               False
grott    |      extname:                 grottext
grott    |      extvar:                  {'none': 'none'}
```
It would be truly amazing if you could take a quick look at this : )

(On another node, the misalignment of reported power draw could be up to the 10 sec vs. 5 min update intervalls of the reporting devices... I will look into this asap)

sgorsh, 2026-01-06:
I'm currently reverse engineering SPF3500 ES + ShineWFI-S, will update when I extract more fields (I don't have solar panels connected yet):
https://gist.github.com/sgorsh/a0a3c6c892eee8bc4713c868ce39c031