johanmeijer/grott #663: Alpha SPH Layout not detected in server mode

Konni1234, 2025-06-01:
Hello in the last 3.2.0 Alpha, are the SPH TL3BH UP Inverters not longer recognized or it detects the incorrect Layout.

Servermode:
`2025-06-01 11:19:07,109 - grottserver -         DEBUG: Main, available threads:
2025-06-01 11:19:07,109 - grottserver -         DEBUG:  - MainThread
2025-06-01 11:19:07,109 - grottserver -         DEBUG:  - httpserver_0.0.0.0:5782
2025-06-01 11:19:07,109 - grottserver -         DEBUG:  - conserver_0.0.0.0:5279
2025-06-01 11:19:07,109 - grottserver -         DEBUG:  - forward_192.168.100.109:59580
2025-06-01 11:19:07,110 - grottserver -         DEBUG:  - client_192.168.40.13:37501
2025-06-01 11:19:07,110 - grottserver -         DEBUG:  - forward_192.168.100.109:58912
2025-06-01 11:19:07,110 - grottserver -         DEBUG:  - client_192.168.40.12:58706
2025-06-01 11:19:07,129 - grottserver -         DEBUG: handle_readble_socket, input received on socket : <socket.socket fd=7, family=2, type=1, proto=0, laddr=('192.168.40.3', 5279), raddr=('192.168.40.12', 58706)>
2025-06-01 11:19:07,129 - grottserver -         DEBUG: start message buffer processing
2025-06-01 11:19:07,129 - grottserver -         DEBUG: Process message buffer split processing
2025-06-01 11:19:07,130 - grottserver -         DEBUG: handle_readble_socket, process data to sent to growatt server
2025-06-01 11:19:07,130 - grottserver -         DEBUG: handle_readble_socket, put data on growatt queue: 192.168.100.109_58912
2025-06-01 11:19:07,130 - grottserver -         DEBUG: handle_readble_socket, data forwarded to growatt server
2025-06-01 11:19:07,130 - grottserver -         DEBUG: process_data, data received from : 192.168.40.12:58706
2025-06-01 11:19:07,131 - grottserver -         DEBUG: 
        \x00\x6b\x00\x06\x03\x3f\x01\x04\x03\x2b\x2b\x47\x23\x33\x23\x77\x46\x5c\x77\x
        61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x3b
        \x27\x2b\x40\x37\x04\x23\x5f\x35\x24\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x
        72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x78\x72\x75\x4c\x61\x6a\x74\x61\x74\x74
        \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x
        74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
        \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x
        6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
        \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x
        74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
        \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x
        77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
        \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x
        47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
        \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x
        61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
        \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x77\xaf\x
        76\x0b\x77\x64\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
        \x47\x72\x6f\x77\x61\x74\x15\x03\x67\x55\x77\x02\x74\x74\x47\x72\x6f\x77\x61\x
        74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x66\x70\x72\xb6\x75\x6d\x77
        \x61\x74\x74\x47\x72\x6f\x77\x8b\xb0\x74\x47\x75\x6b\x71\x90\x73\x76\x47\x72\x
        85\xb3\x61\x74\x75\x4c\x72\x6a\x6c\x1e\x74\x8e\x47\x72\x6f\x77\x61\x74\x74\x47
        \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x06\x74\x46\x40\x8d\x77\x61\x74\x
        d8\x47\x73\x38\xec\x61\x74\x74\xe8\x72\x6c\x19\xa5\x74\x74\x47\x72\x6f\x77\x61
        \x74\x7c\xbb\x72\x6f\x77\x61\x74\x74\x4f\x96\x6f\x77\x61\x74\x74\x47\x7a\x95\x
        77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\xee\x77\x61\x74\x74\x47\x11
        \x7a\x4d\x60\xcc\x75\x17\x72\x5d\x65\x8c\x65\xb2\x47\x72\x28\x23\x60\x1a\x74\x
        1b\x64\x5f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6d\x77\x61\x74
        \x74\x47\x72\x79\x47\x73\x04\x79\x0f\x7f\x51\x7d\xf8\x75\x68\x46\x79\x6f\x87\x
        60\x77\x7e\x62\x72\xcf\x77\x6b\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x76\x0a
        \x73\x80\x74\x36\x47\x72\x1a\x34\x61\x74\x74\x47\x71\x87\x77\x61\x74\x74\x47\x
        72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\xce\x74\x77\x29\xb6\x6f\x77\x61\xdb\x74
        \x44\x1c\xab\x77\x61\x99\x68\x47\x72\x82\x6b\x61\x74\x74\xf5\x72\x6e\x77\x6c\x
        e3\xa5\x47\x73\x6f\x79\xdb\x2b\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
        \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xd6\x61\x74\x74\x47\x72\x
        6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
        \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x76\x48\x70\x46\x64\x
        53\x57\x75\x7c\x77\x7d\x74\x10\x47\x11\x6f\x6e\x61\x74\x74\x46\x72\x6f\x77\x61
        \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x
        77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x8e\x74\x47\x72
        \x6f\x77\x61\x74\x74\x24\x7f\x6f\xf7\x61\x74\x74\x47\x72\x6f\x77\x60\x3a\x1c\x
        47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x16\xf0\x74\x74\x47\x72\x6f\x77\x8c\x68
        \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x72\x6f\x2e\x
        0d
2025-06-01 11:19:07,132 - grottserver -         DEBUG: validate data record
2025-06-01 11:19:07,134 - grottserver -         DEBUG: process_data, plain record: 
2025-06-01 11:19:07,135 - grottserver -         DEBUG: 
        006b0006033f0104445944304247573034330000000000000000000000000000000000000000545
        04a3443435130424500000000000000000000000000000000000000001906010b13050300000000
        0000000000000000000000000000000000000000000000000000000000000000000000000000000
        0000000000000000000000000000000000000000000000000000000000000000000000000000000
[... 422 lines cut ...]
                         "pactogridtot": 0, "plocaloadr": 57600, "plocaloads": 1702, "plocaloadt":
                         113313478, "plocaloadtot": 57600, "ipmtmp": 0, "battemp ": 267,
                         "spdspstatus": 5, "spbusvolt": 7282, "etousertod": 0, "etousertot": 0,
                         "etogridtod": 0, "etogridtot": 0, "edischarge1tod": 114, "edischarge1tot":
                         78562, "eharge1tod": 174, "eharge1tot": 87965, "elocalloadtod": 181,
                         "elocalloadtot": 224970, "dwexportlimitap": 0, "epsfac": 0, "epsvac1": 2303,
                         "epsiac1": 0, "epspac1": 0, "epsvac2": 2276, "epsiac2": 0, "epspac2": 0,
                         "epsvac3": 2299, "epsiac3": 0, "epspac3": 0, "loadpercent": 0, "pf": 0,
                         "bmsstatusold": 0, "bmsstatus": 129, "bmserrorold": 0, "bmserror": 0,
                         "bmssoc": 99, "bmsbatteryvolt": 5449, "bmsbatterycurr": 440,
                         "bmsbatterytemp": 336, "bmsmaxcurr": 50, "bmsgaugerm": 4841, "bmsgaugefcc":
                         4550, "bmsfw": 0, "bmsdeltavolt": 18260, "bmscyclecnt": 366, "bmssoh": 92,
                         "bmsconstantvolt": 5680, "bmswarninfoold": 0, "bmswarninfo": 0,
                         "bmsgaugeiccurr": 0, "bmsmcuversion": 0, "bmsgaugeversion": 0,
                         "bmswgaugefrversionl": 0, "bmswgaugefrversionh": 2, "bmsbmsinfo": 0,
                         "bmspackinfo": 0, "bmsusingcap": 0, "uwMaxCellVolt": 5680, "uwMinCellVolt":
                         4720, "bModuleNum": 3409, "BatNum": 2713, "uwMaxVoltCellNo": 1073,
                         "uwMinVoltCellNo": 267, "uwMaxTemprCell_10T": 240, "uwMaxTemprCellNo": 259,
                         "uwMinTemprCelLNo": 2597, "ProtectpackID": 160, "MaxSOC": 10, "MinSOC": 0,
                         "BMS_Error2": 0, "BMS_Error3": 0, "BMS_WarnInfo2": 0, "ACChargeEnergyTodH":
                         0, "acchargeenergytoday": 66, "acchargeenergytotal": 30019, "acchargepower":
                         0, "70%_invpoweradjust": 1000, "extraacpowertogrid": 0, "eextratoday": 0,
                         "eextratotal": 0, "esystemtoday": 181, "esystemtotal": 224970}}
         - Grott MQTT topic used : energy/growatthalle
`

johanmeijer, 2025-06-07:
I can recreated the problem. But if I use the record from the server mode it also fails in 3.1 and 3.0. 

If I use the record from the proxy mode. It runs fine all versions (3.0, 3.1, 3.2). I am not sure why the first record is not working. Is this an incident or it it a "hard" problem (happens every time)? 

I am have to dig in the records to find the differences, I am afraid.

johanmeijer, 2025-06-07:
Is it also failing in 3.1? That is release where I added some record checking code that might cause a problem (although it worked fine for me).

johanmeijer, 2025-06-08:
I have added some additional debugging code in 3.2.1. maybe this can help with solving this issue. 

If you have time can you run it with this version?

Konni1234, 2025-06-13:
Hello, 

I have done today a new checkout.

Sorry, it doesn't detect the SPHs :(

If you wish, I can give you ssh access to my grott server.

2025-06-13 14:25:23,126 - grottdata -   INFO: Data processing started
2025-06-13 14:25:23,126 - grottdata -   DEBUG: automatic determine data layout started
2025-06-13 14:25:23,127 - grottdata -   DEBUG: Growatt data decrypted V2
2025-06-13 14:25:23,127 - grottdata -   DEBUG: no matching specific record layout found, try generic
2025-06-13 14:25:23,127 - grottdata -   INFO: Record layout used : T060103X
2025-06-13 14:25:23,128 - grottdata -   DEBUG: Original data:
        \x00\x41\x00\x06\x03\x3f\x01\x03\x03\x2b\x2b\x47\x23\x33\x23\x77\x46\x5c\x77\x
        61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x3b
        \x27\x2b\x40\x37\x04\x23\x5f\x35\x24\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x
        72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x78\x72\x79\x49\x6b\x7c\x74\x61\x74\x74
        \x3b\x72\x6e\xde\x29\x74\x74\x47\x16\x6f\x13\x46\x64\x74\x47\x55\x7f\x79\x71\x
        2d\x35\x76\x5c\x5f\x77\x3b\x30\x35\x06\x72\x63\x77\x60\x74\x74\x47\x72\x6f\x31
        \x61\x32\x74\x23\x72\x0b\x77\x61\x20\x24\x0d\x46\x2c\x34\x30\x44\x36\x02\x73\x
        6c\x8f\xc6\x74\x75\x47\x28\x6f\x77\x61\x74\x54\x67\x52\x21\x12\x16\x54\x31\x29
        \x17\x1d\x10\x18\x54\x54\x67\x72\x6e\x79\x70\x74\x74\x40\x9b\x6f\x71\x61\x79\x
        74\x57\x72\x77\x77\x57\x74\x71\x4b\x01\x7c\x03\x73\xfa\x60\x59\x75\x6e\x64\x15
        \x66\xfa\x53\x6c\x6f\x77\x61\x74\x74\x47\x72\x6f\x7a\x5b\x65\x6a\x55\xe1\x7c\x
        fa\x61\xe2\x74\x43\x72\x60\x77\x65\x74\x70\x47\x76\x6f\x72\x61\x71\x74\x47\x72
        \x6f\x77\x61\x74\x74\x56\x6c\x6f\x77\x38\x36\x35\x06\x42\x58\x47\x57\x45\x46\x
        47\x72\x6e\x44\x61\x74\x74\x47\x61\xf3\x77\x49\x64\xd2\x57\x63\x61\xf4\x6e\x6c
        \x74\x53\x72\x6a\x67\x36\x7b\xe4\x47\x16\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x
        61\x74\x7e\x47\x72\x6e\xc3\x61\x8b\x3a\x67\x72\x90\x39\x41\x74\x8b\x09\x52\x6f
        \x88\x2f\x54\x75\x44\x8a\xc8\x76\x62\x75\x77\x47\x72\x6f\x77\x61\x74\x77\xaf\x
        76\x0b\x77\x78\x74\x74\x47\x01\x6f\x45\x60\xa2\x76\x78\x72\x1c\x77\xf0\x26\x73
        \x43\x3e\x6d\xcb\x65\x38\x76\x1f\x72\x6f\x77\x6d\x74\x74\x47\x73\x6f\x77\x61\x
        74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
        \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x71\x
        6f\x74\x61\x71\x74\x42\x72\x0b\x77\x05\x74\x75\x47\x72\x6f\x13\x61\x10\x74\x47
        \x72\x6e\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x63\x74\x
        74\x47\x72\x6f\x77\x61\x75\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
        \x74\x74\x47\x72\x6f\x77\x05\x74\x71\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x
        77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
        \x6f\x77\x61\x74\x74\x47\x72\x0b\x77\x05\x74\x75\x47\x72\x6f\x77\x61\x74\x74\x
        47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
        \x74\x47\x72\x6f\x77\x61\x7b\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x
        61\x74\x74\x47\x72\x6f\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x76\x0a
        \x73\x80\x35\x25\x0d\x47\x5f\x47\x51\x44\x46\x75\x41\x59\x47\x50\x35\x42\x47\x
        72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
        \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x
        74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
        \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x
        6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
        \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x
        74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
        \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x
        77\x61\x74\x74\x47\x72\x95\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
        \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x
        47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
        \x74\x45\x72\x65\x77\xcb\x25\x36\x06\x33\x6f\x7e\x3b\x31\x35\x06\x72\x6a\x16\x
        05 

2025-06-13 14:25:23,128 - grottdata -   DEBUG: Decrypted data:
        00410006033f0103445944304247573034330000000000000000000000000000000000000000545
        04a34434351304245000000000000000000000000000000000000000019060d0e1913030000007c
        0001a9480000006400642710000027100e105941312e30005a444141000c0001000000000046004
        600640064000054504a344343513042450103f8a70001005a000000002020204e657720456e6572
        677920202000010e11000007e90006000d00100018003600050c731374128e141e07011374128e1
        41e00000000000000000d3a111e1293138d00960004000f00040004000400050005000000000000
        0000111e0000594241413037303631320000013300000000139c002810a610110e830f180014000
        510570f90006400000000000000000000000a000001b400ff4e2000ff4e2000ff4e2000ff4e2001
        03f8a70103010300000000000003e80464001900000073003201d6023f007300915207044c02bc0
        44c02580000000c0000000100000000000000000000000000000000000000000000000000000000
[... 142 lines cut ...]
2025-06-13 14:25:23,415 - grottserver -         DEBUG: Process message buffer split processing
[... cut ...]
2025-06-13 14:25:24,412 - grottserver -         DEBUG: Process message buffer split processing
2025-06-13 14:25:24,412 - grottserver -         DEBUG: handle_readble_socket, process data to sent to growatt server
2025-06-13 14:25:24,412 - grottserver -         DEBUG: handle_readble_socket, put data on growatt queue: 192.168.100.109_40606
2025-06-13 14:25:24,412 - grottserver -         DEBUG: handle_readble_socket, data forwarded to growatt server
2025-06-13 14:25:24,412 - grottserver -         DEBUG: process_data, data received from : 192.168.40.13:53010

Konni1234, 2025-06-15:
Hello I send you the logs