johanmeijer/grott #667: record 5129 not processed and a lot of 537 blocked

ZorkaTheDog, 2025-06-20:
Today I realised that I no longer can see my electricity meter and battery info on the growatt website.
Looking at the logs from grott, I see a lot of blocked and not processed messages. I'm not sure when it happened (I probably did an update or so, but it's just now the I see it)

The info from the inverter shows correctly on the website.
Is this fixable?

````
[34m-----------------------------------------------------------[0m
[34m Add-on: Grott Alpha branch (3.0)[0m
[34m The Growatt inverter monitor with current HA plugin integrated[0m
[34m-----------------------------------------------------------[0m
[34m Add-on version: 4a9cb96[0m
[32m You are running the latest version of this add-on.[0m
[34m System: Home Assistant OS 15.2  (amd64 / generic-x86-64)[0m
[34m Home Assistant Core: 2025.6.1[0m
[34m Home Assistant Supervisor: 2025.05.5[0m
[34m-----------------------------------------------------------[0m
[34m Please, share the above information when looking for help[0m
[34m or support in, e.g., GitHub, forums or the Discord chat.[0m

````
````
2025-06-20 16:16:53,890 - __main__ - 	INFO: Grott started, version : 3.0.0_20241208
2025-06-20 16:16:53,891 - grottconf - 	INFO: Config Processing Started
2025-06-20 16:16:53,907 - grottconf - 	INFO: Grott Command line parameters processed:
2025-06-20 16:16:53,907 - grottconf - 	INFO: 	- verbose:   		False
2025-06-20 16:16:53,907 - grottconf - 	INFO: 	- loglevel:   		INFO
False
2025-06-20 16:16:53,908 - grottconf - 	INFO: 	- config file:		/config/grott/grott.ini
2025-06-20 16:16:53,908 - grottconf - 	INFO: 	- output file:		<_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
2025-06-20 16:16:53,912 - grottconf - 	INFO: Grott process configuration file
2025-06-20 16:16:53,917 - grottconf - 	INFO: Grott process environmental variables

	Pulled 'gdecrypt=true' from the environment

	Pulled 'gincludeall=true' from the environment

	Pulled 'ginvtype=AUTO' from the environment

	Pulled 'gblockcmd=true' from the environment

	Pulled 'gnomqtt=False' from the environment

	Pulled 'gmqttip=172.30.33.0' from the environment

	Pulled 'gmqttport=1883' from the environment

	Pulled 'gmqttauth=True' from the environment

	Pulled 'gmqttuser=addons' from the environment

	Pulled 'gmqttpassword=eemaquuphi2ii' from the environment
2025-06-20 16:16:53,921 - grottconf - 	INFO: Correct parameter settings if needed
2025-06-20 16:16:53,923 - grottconf - 	INFO: Grottsettings:

2025-06-20 16:16:53,925 - grottconf - 	INFO: _Info:
2025-06-20 16:16:53,925 - grottconf - 	INFO: 	verrel                                   3.0.0_20241208
2025-06-20 16:16:53,925 - grottconf - 	INFO: 	verrelconf                               3.0.0_20241208
2025-06-20 16:16:53,925 - grottconf - 	INFO: 	verreldata                               nky
2025-06-20 16:16:53,926 - grottconf - 	INFO: 	verrelproxy                              nky
2025-06-20 16:16:53,926 - grottconf - 	INFO: 	verrelsniff                              nky
2025-06-20 16:16:53,935 - grottconf - 	INFO: 	verrelserver                             nky
2025-06-20 16:16:53,935 - grottconf - 	INFO: _Hardcoded:
2025-06-20 16:16:53,936 - grottconf - 	INFO: 	datarec                                  ['04', '50']
2025-06-20 16:16:53,936 - grottconf - 	INFO: 	smartmeterrec                            ['1b', '20', '1e']
2025-06-20 16:16:53,936 - grottconf - 	INFO: 	mindatarec                               12
2025-06-20 16:16:53,936 - grottconf - 	INFO: 	inverterid                               automatic
2025-06-20 16:16:53,936 - grottconf - 	INFO: 	outfile                                  sys.stdout
2025-06-20 16:16:53,936 - grottconf - 	INFO: _Generic:
2025-06-20 16:16:53,937 - grottconf - 	INFO: 	verbose             gverbose             True
[... 103 lines cut ...]
	\x78\x6f\x76\x51\x74\x7f\x47\x72\x6f\x7a\x61\x76\x45\x7f\x72\x7b\x77\x69\x33\x
	20\x14\x25\x5f\x47\x51\x44\x74\x52\x72\x68\x40\x4f\x40\x5a\x76\x5c\x5c\x77\x77
	\x74\x70\x11\x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61\x75\x44\x47\x6c\x6f\x
	72\x26\x39\x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5a\x5a\x51\x42\x59\x75\x42
	\x4f\x46\x57\x4e\x45\x70\x48\x5a\x41\x11\x42 

2025-06-20 16:18:00,756 - grottdata - 	DEBUG: Decrypted data:
	0090000600ed51294e414341434c363054550000000000000000000000000000000000000000415
	a4534434c3630545500000000000000000000000000000000000000004d4d503045364e30385200
	0000000000000000000000000000000000000013ec0004000131000500013100060002333200080
	01e415a4534434c3630545500000000000000000000000000000000000000000009000130000a00
	0130000b0000000d0002313800140008475453573030303000150007372e342e312e33001600045
	6312e30001b000136001c000130001e0005474d542b38001f0013323032352d30362d3230203136
	3a31373a35367036 

2025-06-20 16:18:00,758 - grottdata - 	DEBUG: Grott data ack record or data record not defined, no processing done
````
record 5137
````
2025-06-20 16:19:32,940 - grottproxy - 	DEBUG: Growatt packet received:
2025-06-20 16:19:32,941 - grottproxy - 	DEBUG:  - <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.3', 41532), raddr=('47.254.130.145', 5279)>
2025-06-20 16:19:32,943 - grottproxy - 	DEBUG: Data record validation started
2025-06-20 16:19:32,946 - grottproxy - 	DEBUG: Command block checking started
2025-06-20 16:19:32,947 - grottproxy - 	INFO: Record blocked: 5137
2025-06-20 16:19:32,951 - grottdata - 	DEBUG: Growatt data decrypted V2

johanmeijer, 2025-06-20:
It should also work with blockcmd is true.
With the new inverters and capabilities, we have to adjust the whitelist, probably.

It is possible to add an external whitelist but that will be a challenge with docker and/or ha.

I will look into it when I am back from holidays.

johanmeijer, 2025-06-20:
If you have time you can look at :

recwl.txt in the examples folder.

Copy this one into the grott home directory and add the records that you think need to be sent to growatt.

I do understand to add (reference) this  in docker requires advanced skills.

ZorkaTheDog, 2025-06-20:
Adding recwl.txt to the config dir and adding 5137 to it, solved the batterij not being seen on the Growatt server. But resulteert in a lot of blocked 5138 messages. So I adder that one to, but this doesn’t seem to make a differents, other that not rilling the log with blocked massages.

johanmeijer, 2025-06-21:
Euh.... "You got to love autocorrect"