johanmeijer/grott #669: It's not working after Shine Wi-Fi update to 3.1.1.4

paralichko, 2025-06-22:
Hello,

i am using grott stable 2.8. It was working well until growat updated my shine Wi-Fi to FW:3.1.1.4 and activated ModBus over TCP. Now there's no data in HA.

Debug log attached.

Regards
Gabriel

[grott.log](https://github.com/user-attachments/files/20853612/grott.log)

TheKrowi, 2025-07-09:
Looks like https://github.com/muppet3000/homeassistant-grott is not processing the data correctly as using MQTT explorer outputs this data every minute or so:

`{"device": "ADJ0BGA02F", "time": "2025-07-09T19:59:04", "buffered": "no", "values": {"datalogserial": "XGD5BGT56T", "pvserial": "ADJ0BGA02F", "pvstatus": 1, "pvpowerin": 5577, "pv1voltage": 2054, "pv1current": 13, "pv1watt": 2705, "pv2voltage": 2058, "pv2current": 13, "pv2watt": 2872, "pvpowerout": 5003, "pvfrequentie": 5003, "pvgridvoltage": 2242, "pvgridcurrent": 23, "pvgridpower": 5004, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 82731253, "pvenergytoday": 264, "pvenergytotal": 127228, "epvtotal": 137499, "epv1today": 142, "epv1total": 67813, "epv2today": 143, "epv2total": 69686, "pvtemperature": 341, "pvipmtemperature": 413}}`

peters59, 2025-08-11:
Yes, for me too. Sometimes it works all day, then I can't see data in Home Assistant for days.
At first, I thought restarting the "Growatt 2.8" plugin would help, but that was probably just a coincidence.